Informaticore/SuperPower #24: PCB default config without soldering

wassfila, 2020-10-11:
In the current status and PR https://github.com/Informaticore/SuperPower/pull/23 the user of the PCB would have to solder the 1 cell jumper before he can use the board.
As 95% of users might be using a single cell, suggestion is to make it the default

- the pcb should be operational for 1 cell out of the box
- undersoldering an soldering 0 ohm and bridges should only be required if a different config 2 or more cells are required

wassfila, 2020-10-11:
@MantaRayDeeJay I see two 0 ohm resistors, but is it possible to configure that only one of them is mounted by default ?
[image]

MantaRayDeeJay, 2020-10-11:
[image]
Please see "Add default 1 cell setting"commit

wassfila, 2020-10-11:
I can't see such commit, maybe you did not push it yet ? when you mention a commit, you can copy and paste the hash number and github will create an automatic link, very convenient when we talk about commits

[image]

c1945163cbadd14c11eed62c1720836f3af2eb88

[image]

MantaRayDeeJay, 2020-10-11:
> 
> 
> @MantaRayDeeJay I see two 0 ohm resistors, but is it possible to configure that only one of them is mounted by default ?
> [image]

[image]
As you can see in the table, the 2 pins (CELLS0 and CELLS1) should be tied to INTVcc to select "1 CELL"

wassfila, 2020-10-11:
issue already fixed with c915a490a23e91a1651dea954c5cda66bbbb2c43

closed 2020-10-11 by wassfila