Informaticore/SuperPower #27: generate svg schematics for low cost variant

wassfila, 2020-10-11:
The low cost variant might be our best seller to feed R&D of the Original (Premium) variant, so we might want to update the github action to generate the svg schematics separately.
@Informaticore I remember you mentioned that, but I can't see that in any branch, so we can use this as a reminder and close it simply if it was already done, then maybe link where.

current master 6b6abbf4b22ff191eef2cfc5b6933118083df25d
[image]

Informaticore, 2020-11-09:
this was done last week and can be found in the .generated folder

closed 2020-11-09 by Informaticore

wassfila, 2020-11-09:
@Informaticore is it now generated automatically on each action ? Or do we abandon the idea of automatic generation and update them manually when we need ? That would be ok for me, and if so, the ticket can stay closed.

Informaticore, 2020-11-10:
back-pushing of the generated files is causing too much trouble. The action is still running on push (if *.sch files changed) and PRs so we can always find the files in the artifacts to push them manually.