Informaticore/SuperPower #39: original mcu review

wassfila, 2020-10-20:
Reviewer : Raspi Team : Seth K
here are a couple easy steps that can make it easier to read.
First, I'd recommend using SOME net labels on page
1.  This would reduce the wiring clutter, and make the more important signals stand out.
2. I would recommend adding expanding each circuit block so the circuits have more room to "breathe", and adding a text label under each block to indicate what they are/do.

2010019970909, 2020-10-27:
https://github.com/Informaticore/SuperPower/pull/48/commits/82af7b3988ff3aaf0015fdd3ca698e6752f5edce, adds comments to describe the function of most of the blocks.

wassfila, 2020-11-16:
Feedback from team meeting review notes from @Informaticore 
Meeting Notes:
# Schematic review to da max
    - freeze schematics after last change for regulators
    - create ticket and reviews/findings from everybody there
    - https://github.com/Informaticore/SuperPower/issues/39
    - check requirements against the current schematics if they are met

# Components annotation for BOM
- use same unit for components (100nF vs. 0.1uF)
- deciding for footprints
    - Resistor: 0603
    - avoid BGA
    - capacitor X7R rating
        - <10uF 0603
        - '>10uF 0805
    - wisley choose inductors
- set the foot prints
- swap components by multible different values to reduce component number
- one person should do that cleanup/setting

ToDo:
- put LED indicator for Power on VIN on the charger sheet

wassfila, 2020-11-16:
- check again if the +5V reg should not get a Mosfet instead of driving the Enable pin
uA current won tbd.