Informaticore/SuperPower #43: Finalize MCU branch - DRC error

wassfila, 2020-10-25:
Yep, any help is welcome.
I've tried to track the changes that have still to be done [here](https://github.com/Informaticore/SuperPower/projects/4).

For the DRC, on the MCU sheet, there is only one relative to the MCU (one pin of the pins header not connected, which is related to the replication of the low cost version which I did not do yet). The rest of the DRC errors is on other sheets. Also there was two remarks on #39 (one about using net label (I suppose for the voltages, etc., but they are global one in KiCad, so we have to be in sync).

_Originally posted by @2010019970909 in https://github.com/Informaticore/SuperPower/pull/41#issuecomment-716181248_

wassfila, 2020-10-25:
could not figure out what the error was in the team meeting #42 
[image]

2010019970909, 2020-10-25:
Might be related to https://github.com/nerdyscout/kicad-exports/issues/8. I'll let @Informaticore, if he has the time, get in touch with Nerdyscout to investigate the issue (the warning might be causing the issue though).

2010019970909, 2020-10-29:
Closed by #47, the path separator used by the `update_sym-lib-table.py` script was causing the GitHub action to fail (since UNIX path are used in the KiCad export GitHub action).

closed 2020-10-29 by 2010019970909