Informaticore/SuperPower #44: Charge Status LED

wassfila, 2020-10-25:
one comparator to be used to switch an LED on for status
- monitoring only when USB is connected
- when the current is below 25 mA it will turn the LED on
- to use a comparator (OpAmp) designed to be a comparator and not a generic OpAmp
- impact of sense resistor on the LTC4162 might affect it if its ground is lifted up, so might consider to measure from the top instead of from the ground.

[image]

wassfila, 2020-11-04:
@2010019970909 this might completely change the scope of this ticket, would you prefer to keep this one and we create a new one for the RGB LED ? Or is it ok to rename it like I did ?

2010019970909, 2020-11-04:
@wassfila, maybe it's better to rename this ticket, "charge indicators (measurement and user interface)". IDK.

wassfila, 2020-11-04:
@2010019970909 I did rename it like that. But actually the measurement part is going to fly away if we use the user LED and LTC readings.

wassfila, 2020-11-13:
fixed by #52 
the duplicated charge measure external to ltc won't be needed.

closed 2020-11-13 by wassfila