Informaticore/SuperPower #56: PCB routing Process and first step

wassfila, 2020-11-16:
notes from @Informaticore Team meeting minutes :
# How are we doing the routing?
- try the <click> but then route by hand
- only one branch is used!
- we need a volunteer (most knowledge about kicad?) to start routing. Volunteer decides the time and shares its screen (24h before)
- start routing together with placing components (share ideas)

2010019970909, 2020-11-17:
@wassfila,

- we can use Freeroute (https://youtu.be/BQpAmRV79FM [FR], https://freerouting.org/freerouting/using-with-kicad) for the routing, but it's better to first route the sensitive tracks;
- OK for one branch, but we can make different branches locally to play with the routing tool;
- personally I know how to use it but I'm not an expert (saturday afternoon is a good moment to route some tracks, but first we have to place the components and to define the mechanical dimensions, the rest is usual business xD).

wassfila, 2020-11-18:
@2010019970909 I created this ticket out of the meeting minutes, but anyway we agreed to start this all together, we might have to agree on discord when to have this meeting. if no suggestion comes up, it might end up being next Sunday 19h during the Team meeting.
@Informaticore or should we plan it separately ?
From what I understood, the routing is not that of a big deal that requires automatic routing, but maybe some routing assistance features might make the manual routing smarter and faster.
For branching, as long as it's local, it of course does not matter, only one routing branch will be merged back so to say which will be the commonly used one.
Also we might have to coordinate the routing with regards to the review cleanup, on what should come first. Given that any cleanup in the schematics result in reworking the routing as well, it might be that we shift routing till the "clean freeze" of the schematics :)