Informaticore/SuperPower #57: SuperPower-uC Original Review

sethkaz, 2020-11-16:
commit: 7201a1d
Schematic review only. 

Fix:
- I might be wrong on this, but I believe the power ORing circuit might not work as intended.  
  - I believe the diodes should be either Zener diodes or should be put in backwards.  The intention is probably zener diodes.  However, one must balance the turn-on of the FET with the max voltage being backfed into the lower power supply.  
- There are several capacitors of identical value that have slightly different footprints.  C707&C717, C709&C71, C713&C718.
  - This may seem like a trivial distinction, especially as the difference is on the distance between the pads, but keeping parts consistent will help with making this PCB easier to build and assemble.
- What is the purpose of R4/R5 having both JP1/JP2?  It seems that JP1/JP2 satisfies the need for tying the CELLS0/1 signals to their appropriate nodes.
- The reference designators should follow a consistent pattern.  It should be incremental from 1, such as on the charger page, or it should include the page number, like on the regulator page.  
  - When a design has both, it just ends up being confusing and asking for people to say "what happened here??".  Both systems have their pros and cons, so just pick one and stick with it. 
- The feedback dividers should have one of their resistors be a re-used part.  There is no reason for R701, R703, R706, and R707 to be unique parts.  130k and 150k could easily be changed to 100k, and the top resistor could be changed to another value.  This would eliminate 2 unique parts. 
- For U701 and U702, re-draw the part to fix pins 12/13 and 7/8.  Showing them as "1213" and "78" makes it look like it's a 1213-pin part, rather than the 15 pin part it is.  (Yes, I realize that the package designer at TI got quite creative with the merged, u-shaped pins in the package, but that's not a reason to confuse the reader of your schematic.)
  - Option 1: Use KiCAD's hidden pins to stack pins 13 and 8.  https://kicad.org/libraries/klc/S4.3/  (Just be cautious of the Power Input and Power Output pin designation, as that will make connections you weren't expecting.)
  - Option 2: Show individual pins for 7, 8, 12, and 13.  This adds a little bit of clutter, but does capture all the connections. (Not my preferred solution here.)
  - While you are changing this, you should move FB over to the right hand side, near VOUT, as those two pins are directly related.  
- Use consistent units: C11 (0.1uF) and C1 (100nF).  (My personal preference is to not use nF, but it's not a rigid rule. )
- For R20 & R21, change the values to reduce the unique part count by 1.  The 22k can be changed to 10k, and the 47 can be something around 21.3k (=47k/22k)

You really should Fix:
- Label R12 or Q4A as DNP.  (As a global comment: make the DNP parts clear that they are intended to be DNP.)
- Consider removing R13.  
- The feedback dividers should look like typical feedback dividers in the totem-pole orientation.  
- Rotate the VIN labels on pins U701.1213 and U703.6 so that they connect with the end of the wire.  This will eliminate the hanging wire squares that are visible.  This also applies to U701.1.  
- For R24 and R25: instead of trying to stuff the value into the body of the symbol, it is perfectly acceptable to place the reference designator and value on the wire (in the same sense as the net label) immediately before and after the resistor (or inductor or capacitor).  
- On the MCU module page, VIN is not used on that page. I would recommend removing the wire, and the hierarchical label, along with the wire on the top sheet. 
- I believe the note "Permanent Power Jumper" should be on the charging module block.  This is the circuit R14/Q4A, correct?

Consider fixing:
- The label for the PWR_FLAG being visible doesn't really add anything to the schematic.  Consider hiding those to clean up the schematics.
- Consider changing R1 to be 470 Ohm.  This will eliminate a unique part on the BOM, and is within the suggested range on the datasheet.
- Consider changing R19 an R22 to be 470 Ohm.  This will eliminate another unique part.  The LEDs will still be very much functional with the lower current.  
- Consider hiding the value field for JP1/JP2.  They don't add any clarity to the schematic, but do add clutter. 
- R11 and R7 can probably be unified to a single value to eliminate a unique part. 
- I would recommend re-orienting Q701 to highlight the fact that the voltage is being sourced by 3v3_RTC.  At the moment, it is not clear what it's purpose is. 
- If possible, consider changing Q5 and Q6 to be NMOS instead of an NPN.  This would help reduce part count, and possibly simplify your design.  Alternatively, maybe consider using logic gates to generate the signals needed.  I haven't worked it out, but it looks like 2 XOR gates might get you the same functionality.  
- For nets CHREN, CHR0, and CHR1, you should remove the wire and net label, and leave the no connection flag.  If the pin is not being connected to anything else, then labelling it like that is not necessary and potentially confusing.  

Comments:
- On the regulator page, the package sizes of the capacitors are shown on the sheet.  I would recommend making that a meta-data field of the component, rather than an unattached text.  This would help keep the data together. It would also be easier to move, copy, modify, or hide.  
- Thank you for including a truth table for the autoflash/reset circuit.
- My repeated mentioning of reducing the unique part count is to drive the cost of this board down.  Every unique part is one more reel and feeder that the assembly house needs to put on their pick n place, which means more labor hours ($$$) and more opportunities for mistakes ($$$).  
- It's a nice touch to include all the math and calculations on the schematic.  It's a bit overkill for me, as I would have put those in the sphinx docs, but still the effort to document is appreciated.

MantaRayDeeJay, 2020-11-16:
What is the purpose of R4/R5 having both JP1/JP2? It seems that JP1/JP2 satisfies the need for tying the CELLS0/1 signals to their appropriate nodes.
**ANSWER** : Using only 1 cell is the default setting option.
- In case where the board is bought with components soldered, R4 and R5 should be populated. (No need to add hand solder on "solder-jumpers") -> Just Buy it and use it !
- In case where the user (advanced user) want to use more than 1 cell, the "solder-jumpers" can be soldered manually with the right settings (Removing R4/R5 will be also needed in case the board is bought with components soldered)

MantaRayDeeJay, 2020-11-16:
Consider removing R13.
**Answer** : The divider bridge formed by R10 and R13 initially protect the gate of MOSFET from OVER VOLTAGE.
As the LTC4162L allow 35V on Input, often gate voltage for 30V P-MOSFETs is limited to +/-20V.
But to fully use the onboard components (LDO and Buck-Boost regulators), +15V is the max allowed voltage, so the R13 can not be populated.

MantaRayDeeJay, 2020-11-16:
Consider changing R1 to be 470 Ohm. This will eliminate a unique part on the BOM, and is within the suggested range on the datasheet.
**Answer** : The 9101C chip datasheet tells that we can use a resistor value from 330 to 470 ohms.
So we can use 330, 390 or 470 ohms as standard values.
So, you're right, we can reduce resistors values range !
https://www.diodes.com/assets/Datasheets/AP9101C.pdf
[image]

MantaRayDeeJay, 2020-11-17:
R11 and R7 can probably be unified to a single value to eliminate a unique part.
**Answer** : I'm not agree. 
- R7 is used to limit the overall input current.
- R11 is used to limit only the battery charging current.
So, their functions are not the same !

dr-stefan-krank, 2020-11-17:
Some answers to questions regarding RTC and regulators:

> commit: [7201a1d](https://github.com/Informaticore/SuperPower/commit/7201a1da604b783479e6c57aa4eaa46b97019563)
>     * The feedback dividers should have one of their resistors be a re-used part.  There is no reason for R701, R703, R706, and R707 to be unique parts.  130k and 150k could easily be changed to 100k, and the top resistor could be changed to another value.  This would eliminate 2 unique parts.

These resistor values are taken from the datasheet to select the desired voltage. They may be replaced like you said to get component use down.
This is something that should be done when we optimize the BOM and see what other values we have at hand.

> * For U701 and U702, re-draw the part to fix pins 12/13 and 7/8.  Showing them as "1213" and "78" makes it look like it's a 1213-pin part, rather than the 15 pin part it is.  (Yes, I realize that [...]

These symbols were directly taken from SnapEDA. Yes, they should be reworked. They are ugly.

>     * On the regulator page, the package sizes of the capacitors are shown on the sheet.  I would recommend making that a meta-data field of the component, rather than an unattached text.  This would help keep the data together. It would also be easier to move, copy, modify, or hide.

These sizes are written down there, just because the datasheet listed them there like that even with these exact values. That can be removed later, but I wanted to make sure that this isn't accidentally overseen and poorly chosen.
Especially if some values are the same, but the package is different. I'm no expert enough to decide to chose other packages than the datasheet tells us to use.

> ### RTC
> 
>     * what exactly is the purpose of the J601 in the RTC sheet? Should we add a button here?

Like the annotation says, there should be an exposed pin. This will manually wake up the board if something external happens.
Can be a button, a reed switch for a window or any other type that creates a rising or falling edge.
We could add a button on the board, which would be quiet handy, but we definitely need to expose a pin.

> 
> ### Regulator
> 
>     * package informations is just clutter in the schematics, this information will be in the BOM anyway and can be removed

Should be removed as soon as the BOM is fixed. As explained above, these sizes may be important as they are listed in the datasheet. It's just a very visible reminder.

sethkaz, 2020-11-17:
So just to clear up a choice in words:  The phrase "power oring" is not powering, nor power o-ring.  It is "Power OR-ing".  As in, it is selecting power from A **OR** B.

2010019970909, 2020-11-17:
> So just to clear up a choice in words: The phrase "power oring" is not powering, nor power o-ring. It is "Power OR-ing". As in, it is selecting power from A **OR** B.

Good catch. I don't know why I wrote oring... BTW, for the VIN on the MCU sheet, at one moment I was thinking about adding it to the main pins headers of the board. It's often done on other boards, but it can be removed if it's not pertinent. Personnally for the IO2, 13 and 15, I'd connect them to the pin headers too. They are still IO and it's easier for the troubleshooting to have them mapped too.

MantaRayDeeJay, 2020-11-17:
I believe the note "Permanent Power Jumper" should be on the charging module block. This is the circuit R14/Q4A, correct?
**Answer** : Do you mean about R12/Q4A ?
If yes, there is 2 "populated" options :
- Without "optimize quiescent current in shipmode" option : populate ONLY R12 (Default settings)
- With the Option : populate R10, R13, Q4A and Q3
Rem. : Even if the option is populated, adding R12 will disable the feature and is NON DESTRUCTIVE (MOSFET By-Pass)

MantaRayDeeJay, 2020-11-18:
**Charger** - 390k R13 should be after Q3 and not in between Q4 and Q3!
**Answer** : Could you tell me why ?

Informaticore, 2020-11-18:
> **Charger** - 390k R13 should be after Q3 and not in between Q4 and Q3!
> **Answer** : Could you tell me why ?

because it is like this in the schematics. I might be wrong.
[image]

MantaRayDeeJay, 2020-11-18:
> 
> 
> > **Charger** - 390k R13 should be after Q3 and not in between Q4 and Q3!
> > **Answer** : Could you tell me why ?
> 
> because it is like this in the schematics. I might be wrong. Not sure if it makes a difference tho :)
> [image]

Yes, this is slightly different.
As said in the datasheet, "**RA  and  RB  form  a  voltage  divider  preventing full gate drive to MP1.**" so the position of R13 has no importance when Q3 is drived : the gate is fully protected in both case.
I have changed it because I think it might be an error :
If the voltage on RB (R13) is too high (more than 2.5V - VGSth) the MOSFET can't be drived.
The voltage on RB (R13) is about 0.28 Vout (Vgs max if often +/- 20V for 30V MOSFET as example)
So even with only 5V on Vin, this is not sure to work properly with the "datasheet schematic".

MantaRayDeeJay, 2020-11-19:
what is that resistor for R3 when the battery has an internal protection?
**Answer** : There is 2 "populate" ways :
- With Battery protection feature : U1, Q1, R1 and C1 populated
- Without Battery protection feature : R3 populated
Rem. : If Battery protection feature is populated, populating R3 is also possible (Q1 By-pass).
But U1 will always stay active. (always 3µA quiescent current)

MantaRayDeeJay, 2020-11-19:
SMBALTER LED Resistor has no value
**Answer** : This is normal, the schematic was not finished.
The R value depends of the Forward voltage and Efficiency of the LED which will be used.
Rem. : Forward Voltage depends of the "color" of the LED
I would recommend to use a 0805/0603 package and a high efficiency to have the lower LED current.

MantaRayDeeJay, 2020-11-19:
solder jumper part looks quite complicated do to its symbols, maybe we choose a "easier to read" symbol which creates less track-crossings
**Answer** : In my opinion, this is not specialy complicated.
What would you suggest ?

MantaRayDeeJay, 2020-11-19:
move all images in this sheet to ReadTheDocs on a separate page to explain the settings a bit more detailed. Image do not fit in the svg or pdf format because they do not scale.
**Answer** : Having images in ReadTheDocs is of course welcomed !
Currently, KiCAD is limited and doesn't allow inserting of SVG images in schematic sheets.
Concerning Formulas, I think this is also a good thing leaving them on shematic sheets. I will recommend only to change "Formula images" by some text.
Concerning other images (like curves), we can leave them only on ReadTheDocs

MantaRayDeeJay, 2020-11-19:
SMBALERT might be usefull for the MCU as well, at least add a TP to it. We might wake the MCU when SMBALTER interrupts. (Default is no alert, so it is programmable)
**Answer** : I also think that it would be a nice feature.

MantaRayDeeJay, 2020-11-19:
Battery protection sheet, charger sheet and the charger i/O sheet having an connector for the battery. On the Charger I/O it is marked wrong with V_CHARGE as an input. This can be removed but it uses the S2B-PH-SM4-TB(LF)(SN) connector as a label which is commonly used for batteries, we should keep it. I also would remove the connector from the charger sheet and only keep it on the battery protection sheet. We might then just add 2 pads as well for soldering a battery if wanted.
**Answer** : Don't mix all connector features :
- Battery Protection sheet : The Battery connector is for external battery usage (other than the 18650 embeded)
- Charger sheet : The battery connector is used to power external device/board directly from the battery voltage.
- I/O sheet : It seems there is no battery connector.

Rem. : It was previously talked on the chat !

MantaRayDeeJay, 2020-11-19:
The charger I/O sheet should only get solder pads + barrel jack and no connector or screw-terminal. (maybe screw-terminal but they are big and ugly and expensive)
**Answer** : I'm not agree.
- Screw-terminal is a common connector used on lot of boards. In my opinion, this is a usefull option.
If you have a look on Aliexpress, you will see that this kind of connectors are cheap.
- In case where you'd rather like to use only solder pads, just don't populate Screw-terminal and use their solder pads instead.
This solution will allow to save space on the board and therefore, you can remove the extra solder pads (represented by the JST PH connector)

Informaticore, 2020-11-20:
> what is that resistor for R3 when the battery has an internal protection?
> **Answer** : There is 2 "populate" ways :
>
> * With Battery protection feature : U1, Q1, R1 and C1 populated
> * Without Battery [...]

okay - gotcha! I looked at it again and I see it now, thanks

Informaticore, 2020-11-20:
> move all images in this sheet to ReadTheDocs on a separate page to explain the settings a bit more detailed. Image do not fit in the svg or pdf format because they do not scale.
> **Answer** : Having [...]

of course - if we it is just a formula we should leave it but as text. and curves and things go into ReadTheDocs. We can also have a whole section for calculations and formulars on RTD which we can link in the schematics (just hard to type because I think there are no links :) )

Informaticore, 2020-11-20:
> Battery protection sheet, charger sheet and the charger i/O sheet having an connector for the battery. On the Charger I/O it is marked wrong with V_CHARGE as an input. This can be removed but it [...]

the JST PH connector is the one we should use for the battery. If you check Adafruit you can get lots of batteries there with that connector. The I/O sheet has this connector but it was /is a mistake by me. The idea was to have this for the battery.

I might mix things up so I try to clean it:
INPUT: Solar
HOW: via barrel-jack or screw-terminal (unpopulated) which also gives us pads

INPUT/OUTPUT: Battery
HOW: on-board holder, solder-pads, JST PH. 

Would this answer the confusion?

Informaticore, 2020-11-20:
> Charger sheet : The battery connector is used to power external device/board directly from the battery voltage.
this I don't get.

MantaRayDeeJay, 2020-11-21:
> the JST PH connector is the one we should use for the battery. If you check Adafruit you can get lots of batteries there with that connector. The I/O sheet has this connector but it was /is a [...]

**Answer**
I think that we shouldn't use the same connector for both "External Battery connector" (Battery Input) and "Battery Power connector" (battery Output). They are not on the same nets.
Therefore, to avoid confusion, I'd rather like to use one XH (2,5mm picth - 3A max) and one PH (2,00mm pitch - 2A max).
XH should be used for the "Battery Input" in case we want to charge with higher current.