Informaticore/SuperPower #69: Review Layout

Informaticore, 2021-01-19:
- [ ] add version/revision and date

MantaRayDeeJay, 2021-01-22:
I think that we should improve the GND pins position on J9/J10

[image]

The goal is to do +3.3V and GND pins close together and to add a " virtual shield" between I2C signals and USB signals

MantaRayDeeJay, 2021-01-22:
The copper layer 2 should be defined as a "Power Plane"
[image]

MantaRayDeeJay, 2021-01-22:
The Layer 2 is defined as a power plane and should be a TRUE power plane without any tracks.
The GND plane must clean to avoid any "loop" effect.

[image]

On the ESP board, move the track on layer 3 or layer 4.
On the Charger board, move the track on layer 4.

MantaRayDeeJay, 2021-01-22:
I can see only one pin connected to the "+3.3V_i2c" plane shape, what is the utility of this plane shape ?

[image]

MantaRayDeeJay, 2021-01-22:
3v3_I2C plane shape could be optimized. Only draw the area you need. It will increase insulation distance between other power plane shapes.

[image]

MantaRayDeeJay, 2021-01-22:
SilkScreen Layer : Max input Voltage can't be +20V (TPS63070 only accept 16V max). 
PS : I have specified +15V max on the schematic.

[image]
[image]

Informaticore, 2021-01-23:
>  Add Silkscreen RefDes for Connectors

@MantaRayDeeJay they all got naming and description do you really need the ref?

>  Add more "Net Label" to improve the routing process (especially near the DC-DC converters/charger)

I don't understand what you mean with "Net Label", what are you referring to?

>Add one 2pts connector/pads for each ouput voltage (+3.3V, +3.3V_MCU and +5V). We need to have the possibility to power sensors, external boards, display, etc. (even if the ESP board is connected)

I will try to add some

MantaRayDeeJay, 2021-01-23:
@Informaticore 

    Add more "Net Label" to improve the routing process (especially near the DC-DC converters/charger)

I don't understand what you mean with "Net Label", what are you referring to?

**ANSWER**
An example below (with one of the 2 DC-DC converters).
PS : It should be great to do it for the 2 DC-DC converters and for the LTC4162L charger chip
Then, I could continue the review easily on these 3 critical layouts.

[image]
[image]

Informaticore, 2021-01-23:
okay - got it

MantaRayDeeJay, 2021-01-23:
@Informaticore 

    Add Silkscreen RefDes for Connectors

@MantaRayDeeJay they all got naming and description do you really need the ref?

**ANSWER**
I think mainly about 2pts connectors because there are 4 on the board.
To avoid any confusion, I should recommend to add it. (sometimes, text on silkscreen and text on schematic are not always synchronized). The Ref. Des. stay the only unique ID for each component.

PS : Always think about newbies. (it should be easy for everyone !!!)

MantaRayDeeJay, 2021-01-23:
@Informaticore 

ESP32 module :
The recommanded layout use 12 vias on the Power GND Pad, I can see only one on the board.
I think, vias should be added at the footprint level.

[image]

[image]

[image]

Informaticore, 2021-01-23:
@MantaRayDeeJay do you know how to add vias in a footprint?

MantaRayDeeJay, 2021-01-24:
**ESP32 Module : Antenna Clearance**

I can see that the recommanded clearances are not fully respected.
It should be 15mm min distance (not only 5mm)

[image]

[image]

[image]

⁣
⁣
Possible solution : Use a Mouting Hole without any copper shape and avoid the usage of metal screw (Nylon should be good).

[image]

[image]

Informaticore, 2021-01-24:
We can go back to M2 holes but then we have to go everywhere, also on the charger board.

MantaRayDeeJay, 2021-01-24:
@Informaticore 

We can go back to M2 holes but then we have to go everywhere, also on the charger board.

**ANSWER**
You don't have enough space on the board to use M2.5 Mtg holes (currently 2.5mm hole diameter without enough clearance space from the edge of the board)
Therefore, we haven't any other choice than using M2 holes (2.2mm hole diameter)

PS : We can keep Mtg holes with copper shape for those on the opposite side of the ESP Module Antenna on the charger board.

Informaticore, 2021-01-24:
I made all of them M2.2 (no pad) for consistency

MantaRayDeeJay, 2021-01-24:
@Informaticore 

@MantaRayDeeJay do you know how to add vias in a footprint?

I didn't try it yet.
I think that you could have a look on the footprint used by the LTC4162L to see how it is build (there are vias)

**UPDATE**

Adding Vias in a footprint is the same way as adding padstacks. (all vias pin numbers must be the same as the EPAD padstack)

I have updated the ESP32 module footprint : Epad 3x3 matrix + vias  (as recommanded on the datasheet)
[ESP32-WROOM-32_EPAD_3x3_ThermalVias.zip](https://github.com/Informaticore/SuperPower/files/5862602/ESP32-WROOM-32_EPAD_3x3_ThermalVias.zip)

[image]

[image]

MantaRayDeeJay, 2021-01-24:
@Informaticore 

**Schematic - Max USB Current settings**

Currently, the max current setting for the USB port is 3A.
But the reverse protection diode can sink only 2A max.
Moreover, Max current used by the board is limited to 500mA (LTC4162 settings) to be compliant with USB2 devices.

I think we should limit the USB current to 1.5A for safety.

[image]

MantaRayDeeJay, 2021-01-24:
@Informaticore 

**CP2102N : QFN28 footprint**

I think you should use the "ThermalVias" Version (I think, only 1 via is not enough)
PS : Via hole diameters are defined with 0.2mm. You should redefine them to 0.3mm. (idem as Sethkaz Review)

[image]

[image]

MantaRayDeeJay, 2021-01-26:
@Informaticore 

Power Vdd net is not well routed. (track width = 0.25mm)
You should use wider copper tracks and copper shape

[image]

MantaRayDeeJay, 2021-01-26:
@Informaticore 

Same thing for the "net-(Q3-Pad1)" between MOSFET Q3 and Rsense R13

[image]

MantaRayDeeJay, 2021-01-26:
@Informaticore 

LTC4162-L Layout

Apply the recommended rules from the Datasheet (p. 35)

[image]

DC2038A Demo Board Layout

[image]
[image]
[image]
[image]
[image]

MantaRayDeeJay, 2021-01-27:
@Informaticore 

**Vdd Plane shape issue**

On layer 4, the "Vdd" plane shape is inner the "GND" plane shape.
As you can see, the "GND" plane is generated first and then the "Vdd" plane is not well generated.

The trick is to increase the "Zone Priority Level" feature on the **Filled Zone** properties dialog box.

[image]
[image]
[image]
[image]

Informaticore, 2021-01-27:
> @Informaticore
> 
> LTC4162-L Layout
> 
> Apply the recommended rules from the Datasheet (p. 35)
> 

this will be a bigger refactor and change. I will try to do that.

Informaticore, 2021-01-27:
> @Informaticore
>
> **Vdd Plane shape issue**
>
> On layer 4, the "Vdd" plane shape is inner the "GND" plane shape.
> As you can see, the "GND" plane is generated first and then the "Vdd" plane is not well [...]

I know, I forgot about it but strangely the DRC was not warning about it!! That is very bad, it shows 0 issues. Do you see DRC complaining?

MantaRayDeeJay, 2021-01-27:
@Informaticore

    Vdd Plane shape issue

    On layer 4, the "Vdd" plane shape is inner the "GND" plane shape.
    As you can see, the "GND" plane is generated first and then the "Vdd" plane is not well generated.

    The trick is to increase the "Zone Priority Level" feature on the Filled Zone properties dialog box.

I know, I forgot about it but strangely the DRC was not warning about it!! That is very bad, it shows 0 issues. Do you see DRC complaining?

**ANSWER**

YES, I can see it !!!

[image]
[image]
⁣
By setting "Zone Priority Level" = 1, DRC is clean as you can see below !!!
⁣
[image]
[image]

MantaRayDeeJay, 2021-01-28:
@Informaticore

    LTC4162-L Layout

    Apply the recommended rules from the Datasheet (p. 35)

this will be a bigger refactor and change. I will try to do that.

**NEXT**

This is better but it can be improved. 

[image]
[image]
[image]
[image]
[image]

Informaticore, 2021-01-29:
@MantaRayDeeJay I applied more changes but I am not 100% about you drawings, please describe a bit more what else should be changed. But I think I got most of it, please check. All the thick traces are thick now, should be good.

MantaRayDeeJay, 2021-01-29:
@Informaticore 

@MantaRayDeeJay I applied more changes but I am not 100% about you drawings, please describe a bit more what else should be changed. But I think I got most of it, please check. All the thick traces are thick now, should be good.

**ANSWER**
I would like VBat Capacitor (C6) and Rsense (R20) as the recommanded layout used on the DEMO Board.
By moving J2, you should have enough place to do it.
Moreover, you should keep the RTC chip and Battery protection circuitry enough away from the inductor. (You can also add a shield GND plane around the Inductorr with some Vias on the layer Top)

MantaRayDeeJay, 2021-01-29:
@Informaticore 

**TPS63070 - Footprint Issue**

1) Pin 12-13 copper area is wrong for the soldermask and solderpast layer.

[image]
[image]
[image]

2) The footprint is SMT !!! (not THT)

[image]

3) You must delete any global solderpast and soldermask layers on all copper pads. (The footprint countains custom solderpast and soldermask pads)
-> Check all copper pads
[image]

MantaRayDeeJay, 2021-01-30:
@Informaticore 

**TPS63070 : Apply the recommanded Layout**
1) Place C1 and C4 as Figure 51 (C19 & C24 for U9 and C21 & C25 for U10)
2) Route L1 and L2 nets as Figure 51 (use vias under pads)
3) Add a gnd shield around the inductor on the top layer as Figure 51

[image]
[image]
[image]

MantaRayDeeJay, 2021-01-30:
@Informaticore 

Maybe I found critical signals too near the LTC4162-L inductor without GND shield on top layer.
I recommand to keep away critical signal near inductors.
In this case, it concerns 2 MOSFET gate pin (Q1 and Q5).

[image]

MantaRayDeeJay, 2021-02-01:
@Informaticore 

LTC4162-L : Separate Inductor GND Shield from other GND Tracks.

[image]

MantaRayDeeJay, 2021-02-01:
@Informaticore 

TPS63070 : Separate Inductor GND Shield from other GND Tracks.

[image]

[image]

MantaRayDeeJay, 2021-02-01:
@Informaticore 

TPS63070 : Optimize Inductor Power shapes against EMI noise : Remove some extra shape area to improve clearance from others nets.

[image]

MantaRayDeeJay, 2021-02-01:
@Informaticore 

TPS63070 : Remove "Artifact Tracks" near capacitors

[image]

MantaRayDeeJay, 2021-02-01:
@Informaticore 

**USB Data Lines** : Use a differential pair by following the recommanded rules on the link below.

High Speed USB Design Guidelines
http://ww1.microchip.com/downloads/en/AppNotes/doc7633.pdf

[image]

[image]
[image]

[image]
[image]