Informaticore/SuperPower #70: Review of layout in feature/layouting_square_stack_mirko (5714cf4)

sethkaz, 2021-01-20:
Need to fix:
1.	Fix the DRC issues around the mousebite vias.  You will have to work with the board house as to how they would like to proceed with this.  KiCAD allows 0.41mm diameter vias with a 0.4mm hole, so maybe use that to clear up the DRC.
2.	Fix the DRC issues around the TPS63070 components.  If you make that hidden trace on pin 10 to be 0.2mm wide instead of 0.25mm, it will clear up the error.
3.	There is a sliver of ground plane that will act like an antenna directly adjacent to J7.13 on the F.Cu layer.  Adjust the trace from J7.14 to fix this.
4.	On page 4 of the schematic, J5.1 is connected to +BATT.  However, in actuality, it is connected to the local VBAT label.  I would HIGHLY recommend changing the label on J5.1 to VBAT, because one shouldn’t need to go to 2 other schematic pages to find that two differently named nets are connected. 
5.	The trace between R20.2 and L1.2 is thin and could be widened.
6.	The lack of reference designators will be very troublesome for people who will need to debug this.  Or for people who want to probe the board to see how it works.  We’ve had long discussions about this, but my recommendation doesn’t change:  If you have a silkscreen layer and have the space, you absolutely should include that info in there.  
7.	Pin R9.2: There is no reason for this trace to the plane to be as long or as circuitous as it is.  Please shorten it to a simple drop to plane trace. 
8.	Pin R22.1: There is a via in pad here.  It doesn’t serve a purpose, so remove it.  While you’re at it, I would recommend removing the other extra via in the immediate vicinity as well.
9.	U10 and L3 have some very small traces between them.  The converter is taking 3.6A gulps each time it switches, so those thin traces will start to really make an impact.  I would recommend rotating U10 counterclockwise by 90 degrees to make the routing easier. 
10.	Same goes for U9 and L2.  
11.	The traces around L1 (both pins) are very small.  The narrow traces add to the ESR of the inductor, and reduce the efficiency.  The calculation for the inductor current is 45mv/68mOhm = ~660mA.  It’s not a large current like the 3.6A for U10 and U9, but it’s enough to warrant widening the traces or even better, using copper pours. 
12.	Consider reducing the thermal clearance on the planes.  This will generally improve the coverage in the board.  They are currently set to 0.508mm, which is a very large clearance.  You could easily change this to 0.2mm without any impact on manufacturing.  This is especially a problem under U4 where the vias have only a few spokes, and the center vias are isolated.  This is bad practice for thermal design.  

Should fix:
13.	Pin U4.18:  There’s no reason for there to be 2 vias directly adjacent to this.  Pull the traces on In1.Cu and B.Cu from a single via.  
14.	Change the silkscreen note from “FTDI” to “USB-Serial”, or something similar.  This is because the chip being used is not made by FTDI, so it could lead to some confusion
15.	I’d recommend widening the plane on In2.Cu around the pin J10.10.  There is only one thermal spoke for this pin, and it can easily be fixed.  
16.	Pin Q1.6&7, consider merging this to a single GND via, or at least move them a little further away from each other.  The PCB vendor might complain about the drill hits being too close together.  
17.	The USB C connector might need to be nudged a little away from the edge of the board (moving to the right or East).  This is because the PCB vendor has rules about the distance between copper features and the board edge.  A typical number I’ve seen 0.020” (or 0.508mm).  
18.	The same can be said for the copper of the mounting holes.  
19.	Pad C28.1 has a via very close to it, and could contribute to solder thieving.  

Consider fixing:
20.	The hole size for the LTC4162 is 0.2mm.  This might be a challenge to build at some board houses for a reasonable cost. 
21.	The Silkscreen notes on the right hand board have a thickness of 0.125mm.  This might be a problem for some PCB vendors to make.  I would recommend staying above 0.006” (0.153mm). 
22.	At Q1.3&2, consider moving the trace bridge out from this location.  While it is easier to route, it can cause false alarms in the automated inspection steps.  As in, it looks like a short between pins, and the AOI steps are looking for that.  
23.	I would recommend ditching the GND plane on the F.Cu layer completely. It is only being used in 2 places, and both of those can be replaced with a trace very quickly.  The disadvantage is that it adds a lot of copper features that could be places that act as antennas or worse.  
24.	The connector for J4 is pretty fine in pitch.  If the intention is for people to manually solder something in, maybe consider moving up to a 2mm pitch?  There is plenty of room to do so. 
25.	The trace between R36.1 and Q8.3 can be re-routed to not use the ground plane.  One easy method is to move R36 next to Q7, and rotate it 180.  The track could then go above Q7 for easy routing. 

Notes/Comments:
26.	I like the silkscreen outlines of the functional areas.  That helps explain what’s happening on the PCB, and mitigates my silkscreen comment above.  
27.	So I noticed that the connector J3 has 2 large holes with no electrical contacts, and 2 surface mount pins.  I’m assuming this is a method of strain-relief for the wires, correct?   If so, as the hole size is not very critical, I would recommend reducing the number of unique hole sizes by increasing this diameter to match the mounting hole size.  As in 2.2mm -> 2.5mm.

Informaticore, 2021-01-20:
Is there a global setting for thermal clearance?

Informaticore, 2021-01-20:
15.: widening the plane was not helping to get more thermal spokes. The clearence of the Pads next to it is to large. should I reduce the clearance for that component? Is there an global setting? Or should I leave it then?

sethkaz, 2021-01-20:
> 15.: widening the plane was not helping to get more thermal spokes. The clearence of the Pads next to it is to large. should I reduce the clearance for that component? Is there an global setting? Or should I leave it then?

To edit them globally, use the button in the lower left of the Copper Zone Properties window.  

I was able to get it wrap around and make 3 connections.  You just have to play with all 4 numbers a bit.  

[image]

Informaticore, 2021-01-20:
> 20. The hole size for the LTC4162 is 0.2mm. This might be a challenge to build at some board houses for a reasonable cost.
what hole size would be better?

sethkaz, 2021-01-21:
It really depends on the vendor.  PCBway's "cheapest" hole size is 0.3mm.

Informaticore, 2021-01-21:
>18. The same can be said for the copper of the mounting holes.

Moving them will create a big mess - I will try it out and see if the manufacturer says anything against it :)

Informaticore, 2021-01-21:
>  20. The hole size for the LTC4162 is 0.2mm. This might be a challenge to build at some board houses for a reasonable cost.

I mark that as done - I will leave it as is because that is the original footprint

Informaticore, 2021-01-21:
Closed by mistake, sorry

sethkaz, 2021-01-21:
> > 1. The same can be said for the copper of the mounting holes.
> 
> Moving them will create a ~big~ mess - I will try it out and see if the manufacturer says anything against it :)

Yes, you can try.  An alternative is to just reduce the size of the copper without changing the location.

Informaticore, 2021-01-22:
>  24. The connector for J4 is pretty fine in pitch. If the intention is for people to manually solder something in, maybe consider moving up to a 2mm pitch? There is plenty of room to do so.

we will keep it as is - if it makes trouble in practice we might change it. it is a rare feature. 
With that - I am done with all the topics :)

MantaRayDeeJay, 2021-01-22:
I'm not fully agree about the point n°23 :
"I would recommend ditching the GND plane on the F.Cu layer completely. It is only being used in 2 places, and both of those can be replaced with a trace very quickly. **The disadvantage is that it adds a lot of copper features that could be places that act as antennas or worse.**"

I think that GND plane shape **will act as a shield if linked to the True internal GND Plane via Vias**.
If there is no lot of free space as on the layer 1 (Top), this is not very usefull, we can keep without it.
But I recommend to add it on the layer 4 (Bottom).
 
Moreover, PCB manufacturers don't like very much when there are a lot of copper to remove by the chemical process.  
(It needs to use more "iron perchloride", therefore less ecological)