Informaticore/SuperPower #73: Can't load 3d model with FreeCad

wassfila, 2021-01-25:
commit 43f4432fa330a83b1790b3b61581f4cfa3f51369

file -> Open -> "SuperPower-uC\original\SuperPower-uC-KiCad.kicad_pcb"

Error :
[image]

[image]

[image]

[image]

wassfila, 2021-01-25:
for the raspberry pi
commit 
https://github.com/sethkaz/SuperPower/commit/d3b12cf72331b07d9264cd0fdebfc268e367fad2

[image]

2010019970909, 2021-01-25:
I cannot reproduce the error, but I remember having it a few months ago.

[image]

I only had the info this time:

[image]

MantaRayDeeJay, 2021-01-25:
@wassfila 

Try to use the "KiCAD StepUp" Module ToolBar to open the file.

[image]

MantaRayDeeJay, 2021-01-25:
@wassfila 

Which FreeCAD version are you using ?

wassfila, 2021-01-26:
I have Freecad 0.18 but I can try the latest, I placed the StepUpMod on the Mod folder
[image]

And I only have this menu
[image]

wassfila, 2021-01-26:
I think I was using a "fake" free version of Freecad I got from here

now I'm trying out this, that looks a bit more serious
https://www.freecadweb.org/downloads.php

Is that where you go it from ?

MantaRayDeeJay, 2021-01-26:
@wassfila 

Try with the lastest v0.19 (Conda)
https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre
 
Then use the Add-On Manager from the Tools Menu (last item) to download the "KiCAD StepUp" module.

[image]

wassfila, 2021-01-26:
with developer version 0.19, still the same issue, 

[image]

I must be doing something radically wrong.
Starting from all wiki telling about installing from add-on manager but not telling how

The addon managed of version 0.18 looks like this
[image]

wassfila, 2021-01-26:
@MantaRayDeeJay thanks for your screenshot, that helped, only the version 0.19 show it and I was searching for S "StepUp" which it's in K "KicadStepUp" :)

wassfila, 2021-01-26:
now version 0.19 with StepUp installed from the addon manager still the same issue lol

[image]

I probably need the latest kicad then

MantaRayDeeJay, 2021-01-26:
@wassfila 

The Add-On Manager of the version v0.18.4 is not Up to date.
I saw that they want to release the v0.18.5 (with the add-on Manager Up to date)

But currently, the easiest way to download FreeCAD modules is from the download manager of the v0.19.xxx series.

wassfila, 2021-01-26:
@MantaRayDeeJay are you running windows, cause I'm on windows
still v 0.19 look different
[image]

Can you screenshot your about menu ?

The toolbar is still different
[image]

MantaRayDeeJay, 2021-01-26:
@wassfila 

I can see that on the last error screenshot, that it loads 3D models from "Raspberry" folder.
It could explain the issue.

[image]

Informaticore, 2021-01-26:
I am running 0.18 without the plug-in installed extra. Plain FreeCad and I just press open (folder on top) and open the pcb file. There is an error after loading the PCB but it is just because of something which can not be drawn but I see all things and the pcb is opened.

MantaRayDeeJay, 2021-01-26:
[image]

wassfila, 2021-01-26:
I tried these two versions and they end up on a different About menu info than your screenshot, weird :
[image]
[image]

MantaRayDeeJay, 2021-01-26:
I didn't have the lastest version (v0.19-23546)

wassfila, 2021-01-26:
It's just that Kicad does not like me, might be related to the install path, I installed on D: and the plugins gets installed on the C , if it's that stupid, it would be hard to debug
[image]

MantaRayDeeJay, 2021-01-26:
"C:\Program Files\KiCad\share\kicad\modules\packages3d" is the good path (same as mine)

Could you check KiCAD Footprint paths ?

[image]

wassfila, 2021-01-27:
It worked out, thanks a lot @MantaRayDeeJay it's thanks to your help, I realized that the StepUp Plugin was stupidly assuming that I have kicad installed on C:\ with a hardcoded path, I tried declaring the env var, even patched the hardcoded path in python, nothing helped, I had to copy all models on that C:\ path, then it worked.
It is so far fetched that today programs still do that, but then when I remember that both FreeCad and Kicad are free, I take it back, I used to hardcode paths myself when I was a kid lol

wassfila, 2021-01-27:
[image]

closed 2021-01-27 by wassfila