Informaticore/SuperPower #77: Artemis - Stacked Version - Board Layout Review

MantaRayDeeJay, 2021-02-07:
_Review Status : Work in Progress_

**GLOBAL**

**Layout**
- [ ] Improve ESP32 Antenna clearance area. (See comments on Apollo board Review Ticket)

**DRC Issues**
- [ ] Solve GND MouTinG Holes issue. (Hole and Pad size can't be the same if you plan to connect the Mtg Hole to a net)

---
**DETAILS**

**Layout**

- ESP32 Antenna clearance

[image]

Tobi3566, 2021-02-08:
Some comments on the review points:
ESD: no low cost dev board that i know of has ESD on the usb input and this version is still kinda intended to be lowcost and possible to hand solder

U2: connected the same way basically all charger boards have the mosfet connected (some even have filled plane). also doesn't cause any electrical issues at all

VBat track width: not really possible with the layout and it is also sufficient for the expected current (slightly improved it already)

MantaRayDeeJay, 2021-02-09:
@Tobi3566 
ESD Protection: why and how to protect microcontrollers efficiently
https://www.youtube.com/watch?v=56eZHBBuvsk
[image]