Informaticore/SuperPower #86: Prototype v1.0 Testing Protocol

Informaticore, 2021-04-27:
- [ ] USB-C Delivers 5V when connected
- [ ] USB-UART connection established
- [ ] auto Flashing ESP32
- [ ] enable 3.3V via GPIO
- [ ] enable 5V via GPIO
- [ ] i2C detect LTC
- [ ] i2C detect RTC
- [ ] ... TBD

dr-stefan-krank, 2021-05-02:
Noticeable coil whining on different loads for 3v3 MCU regulator.

MantaRayDeeJay, 2021-05-14:
SOT23-3 DMN3023L MOSFETs correction :
[image]

MantaRayDeeJay, 2021-05-23:
> 
> 
>     * Noticeable coil whining on different loads for 3v3 MCU regulator.
> 

Ceramic capacitor whinings come from a "not good working" of the LTC4162-L chip due to the SOT23-3 DMN3023L MOSFETs issue

MantaRayDeeJay, 2021-07-24:
Ibat Rsense (R20) : 
Currently, the resistor used is a 0805 68mohm 0,25W
Buck-Boost chips can sink more than 2A from the battery.
In this case, vsns = 32mV x4 (128mV) @I=2A -> 0,256W.
There are 2 chips on board, so Power can raise about 0,5W.
I recommand to replace the 0805 0,25W by a 1206 1W resistor.